helgeerbe/picframe #498: "reshuffle_num: 1" setting - did anything change?

sapnho, 2025-07-22:
My wife noticed that the order of the photos is often the same, so I was wondering if my setting of `reshuffle_num` has been incorrect throughout the years.

@paddywwoof will know this, and I apologize for asking an obvious question. :-)

When the setting is `reshuffle_num: 1`, it means that each sequence is only shown once, right?

sapnho, 2025-07-22:
That's really interesting. 

I, of course, argued to my lovely wife that "no, that cannot be," but promised to ask the master.

It would be great to discover that the random function had indeed a bug.

paddywwoof, 2025-07-24:
OK I've done a few tests and checked sqlite behavior. When I restart my RPi it comes up with a different, random, set of pictures each time, and the sqlite documentation bears this out. There is no facility that allows the sqlite random function to be seeded.

I think this shows how good human brains are at spotting patterns, and how bad they are at statistics! If you have `reshuffle_num: 1` it means that a new random order will be generated *every time* all the pictures have been shown. That means that there is a large chance that one of the last 50 images in the previous sequence will be one of the 50 image at the start of the new sequence.* If you have 4,000 image then it's nearly 50% probability that this will happen. Each time you run through without reshuffling there is 0% chance; you will wait 4,000 images before seeing the same image again. Of course, if there is a noteworthy sequence, say a christening followed by a wedding followed by a funeral, then an observer will recognize that the sequence has been repeated, but that's quite unlikely.

Let me know what happens if you restart your pictureframe a few times. If you get the same sequence then there is obviously something going on and something might have changed!

Paddy

* Chance of any picture *not* being in the first `n` out of a total `N` is `(N - n) / N` i.e. quite likely if `N` is much bigger than `n`. However if *all* of the last n pictures are not in the first `n` pictures the pictures after a shuffle that becomes `((N - n) / N) ** n` i.e. (3950 / 4000) ** 50 => 53% or, putting is the other way round 47% chance that a picture *will* reappear in the first 50

sapnho, 2025-07-24:
Yeah, I was afraid that would be the outcome. It's just crazy that with >2000 photos, there are some that appear much more frequently than others.

Thanks for checking!

sapnho, 2025-07-24:
Indeed, I had looked at the database numbers. But since I often add photos or set with date filters, the numbers are not that representative.