helgeerbe/picframe #512: Playing Videos

StonyManager, 2025-07-28:
Videos do not play on my picframe setup. Instead, I get a single image which is apparently created by picframe. See screenshot. 

[image]
Looking through the issues here, there is mention that video playing is experimental and, perhaps, only available on a branch of the picframe github. Could someone please clarify? I'd love to have videos playing. I also ran across a script in one issue that pre-processes videos. Is that useful/required/incorporated? 

Thanks for the clarification/instruction.

helgeerbe, 2025-07-31:
Hi @StonyManager,

video playing should work on the main branch.  And as can see the first and last frame of the video are extracted for the transition blending.

picframe uses the vlc player. Is it installed? Does it play your video files running as a standalone player? Can you provide us the log? In the configuration.yml you could temporarily enable logging to a file.

StonyManager, 2025-08-27:
I got video working! Had to laugh and shake my head, though. It turns out that if you set the configuration.yaml parameters time_delay, show_text_tm, and fade_time in such as way as to make time_delay <= show_text_tm + fade_time, then videos do not play, apparently because they don't have time to get started. The opening frame seems to be displayed with the text indicating it is displaying the video, but it doesn't play and picframe moves on to the next photo/video.

closed 2025-09-26 by github-actions[bot]